ntnu-arl/aerial_gym_simulator #1: RuntimeError: nvrtc: error: invalid value for --gpu-architecture (-arch)

rasrab1992, 2023-11-07:
Hello,

I am a researcher from IMEC involved in the SPEAR project. I'm eager to explore your simulation tool before the kickoff meeting so that I can formulate questions and concerns for WP2. However, I encountered an error while trying to install the 'aerial_gym_simulator,' as shown in the image below:
[image]
I have attempted to find a solution by searching the Nvidia forums and discovered that others have experienced a similar issue related to the RTX 4090 GPU, as documented [here](https://forums.developer.nvidia.com/t/rtx-4090-is-not-woking-with-isaacgymenvs-and-nvrtc-error-invalid-value-for-gpu-architecture-arch/271323). I've tried both conda and docker, but the problem persists.

Could you please provide your suggestions on how to proceed with resolving this issue?
Maybe one solution is changing the driver version to 525?

Thanks

mihirk284, 2023-11-08:
Hi, this seems to be an issue with PyTorch and nvrtc for the new RTX 4090 GPUs. Please check [this issue](https://github.com/pytorch/pytorch/issues/87595), and more specifically, [this comment](https://github.com/pytorch/pytorch/issues/87595#issuecomment-1289832110).

As suggested in the thread, perhaps upgrading a python version to a nightly build may resolve the issue as [detailed in this comment](https://github.com/pytorch/pytorch/issues/87595#issuecomment-1405545201).

rasrab1992, 2023-11-09:
Hello, Thank you for your help. I will test it and let you know about the result.

pompomO, 2023-11-24:
> Hello, Thank you for your help. I will test it and let you know about the result.

Hi, have you solved this problem?

pompomO, 2023-11-24:
here is my nvidia driver version and cuda version
[image]
[image]

EtorArza, 2024-02-16:
I also had this issue, and I was able to solve it with 

```
pip3 install --pre torch torchvision torchaudio -f https://download.pytorch.org/whl/test/cu117/torch_test.html
```

and restarting the computer (as suggested in the comment https://github.com/pytorch/pytorch/issues/87595#issuecomment-1289832110). 

I'm running pop OS 22.04 on a 4080 with cuda_11.5.r11.5/compiler.30672275_0

mihirk284, 2024-02-16:
@pompomO @rasrab1992  Can you please try the above solutions to see if they work?

If not, please try out the fix provided in #3.

MJavadZallaghi, 2024-05-07:
> @pompomO @rasrab1992  Can you please try the above solution [here](https://github.com/ntnu-arl/aerial_gym_simulator/issues/1#issuecomment-1948468678) to see if it works?
> 
> If not, please try out the fix provided in #3.

I'm facing the same issue with GPU RTX 4060 and Ubuntu 20.04, and going to test the solutions provided here. 

I will post the results here after testing your solutions.

MJavadZallaghi, 2024-05-07:
> I'm facing the same issue with GPU RTX 4060 and Ubuntu 20.04, and going to test the solutions provided here.
>
> I will post the results here after testing your solutions.

I have tested [this solution](https://github.com/ntnu-arl/aerial_gym_simulator/issues/1#issuecomment-1948923612), but the issue is not solved.

My terminal's screenshot:
[image]

EtorArza, 2024-05-08:
@MJavadZallaghi what version of torch, torchvision and torchaudio are you using? 

I'm on `torch==2.3.0, torchvision==0.14.1+cu117, torchaudio==0.13.1+cu117`. 

https://github.com/pytorch/pytorch/issues/87595#issuecomment-1865391649 says that updating these to newer versions can solve the issue. You might also need to update the cuda version. Im on `CUDA Version: 12.4`

MJavadZallaghi, 2024-05-08:
@EtorArza This is version list of all the installed package inside the `rlgpu` conda environment (installed by isaac gym preview 4):
```bash
# packages in environment at /home/mjavadzallaghi/anaconda3/envs/rlgpu:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
absl-py                   2.1.0              pyhd8ed1ab_0    conda-forge
aerial-gym                1.0.0                     dev_0    <develop>
aiohttp                   3.7.4.post0      py37h5e8e339_1    conda-forge
async-timeout             3.0.1                   py_1000    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blas                      2.122                       mkl    conda-forge
blas-devel                3.9.0            22_linux64_mkl    conda-forge
blinker                   1.6.3              pyhd8ed1ab_0    conda-forge
brotli-python             1.0.9            py37hd23a5d3_7    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
ca-certificates           2024.3.11            h06a4308_0  
cachetools                5.3.3              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py37h43b0acd_1    conda-forge
chardet                   4.0.0            py37h89c1867_3    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
click                     8.1.3            py37h89c1867_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
cryptography              38.0.2           py37h5994e8b_1    conda-forge
cuda-cudart               12.1.105                      0    nvidia
cuda-cupti                12.1.105                      0    nvidia
cuda-libraries            12.1.0                        0    nvidia
cuda-nvrtc                12.1.105                      0    nvidia
cuda-nvtx                 12.1.105                      0    nvidia
cuda-opencl               12.4.127                      0    nvidia
cuda-runtime              12.1.0                        0    nvidia
cudatoolkit               11.1.1              hb139c0e_13    conda-forge
cycler                    0.11.0                   pypi_0    pypi
ffmpeg                    4.3                  hf484d3e_0    pytorch
fonttools                 4.38.0                   pypi_0    pypi
freetype                  2.12.1               h267a509_2    conda-forge
fvcore                    0.1.5.post20210915            py37    fvcore
gmp                       6.3.0                h59595ed_1    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
google-auth               2.23.0             pyh1a96a4e_0    conda-forge
google-auth-oauthlib      0.4.6              pyhd8ed1ab_0    conda-forge
grpc-cpp                  1.48.1               h30feacc_1    conda-forge
grpcio                    1.48.1           py37he7b19e7_1    conda-forge
icu                       73.2                 h59595ed_0    conda-forge
idna                      3.7                pyhd8ed1ab_0    conda-forge
imageio                   2.31.2                   pypi_0    pypi
importlib-metadata        4.11.4           py37h89c1867_0    conda-forge
intel-openmp              2024.1.0                 pypi_0    pypi
iopath                    0.1.9                      py37    iopath
isaacgym                  1.0rc4                    dev_0    <develop>
jpeg                      9b                   h024ee3a_2  
kiwisolver                1.4.5                    pypi_0    pypi
lame                      3.100             h166bdaf_1003    conda-forge
ld_impl_linux-64          2.40                 h55db66e_0    conda-forge
libabseil                 20220623.0      cxx17_h05df665_6    conda-forge
libblas                   3.9.0            22_linux64_mkl    conda-forge
libcblas                  3.9.0            22_linux64_mkl    conda-forge
libcublas                 12.1.0.26                     0    nvidia
libcufft                  11.0.2.4                      0    nvidia
libcufile                 1.9.1.3                       0    nvidia
libcurand                 10.3.5.147                    0    nvidia
libcusolver               11.4.4.55                     0    nvidia
libcusparse               12.0.2.55                     0    nvidia
libffi                    3.4.2                h7f98852_5    conda-forge
libgcc-ng                 13.2.0               h77fa898_7    conda-forge
libgfortran-ng            13.2.0               h69a702a_7    conda-forge
libgfortran5              13.2.0               hca663fb_7    conda-forge
[... 60 lines cut ...]
sqlite                    3.45.3               h2c6b66d_0    conda-forge
tabulate                  0.9.0              pyhd8ed1ab_1    conda-forge
tbb                       2021.12.0                pypi_0    pypi
tensorboard               2.11.2             pyhd8ed1ab_0    conda-forge
tensorboard-data-server   0.6.1            py37h52d8a92_0  
tensorboard-plugin-wit    1.8.1              pyhd8ed1ab_0    conda-forge
termcolor                 2.3.0              pyhd8ed1ab_0    conda-forge
tk                        8.6.13          noxft_h4845f30_101    conda-forge
[... cut ...]
yacs                      0.1.8              pyhd8ed1ab_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.7.2            py37h540881e_2    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
```
`pytorch` version is 1.8.1. How can I upgrade it inside the `rlgpu` env? Because I'm not sure about dependencies.

EtorArza, 2024-05-08:
@MJavadZallaghi upgrading your pytorch might solve the issue. Activate the environment `conda activate rlgpu` and then follow the instructions in https://pytorch.org/ in section "Install PyTorch". 
The installation command is 

```conda install pytorch torchvision torchaudio pytorch-cuda=12.1 -c pytorch -c nvidia```

However, for the latest pytorch version 2.3.0, it says in the website that python 3.8 is required, and you seem to be running python 3.7, so I don't know if it will work. I think @mihirk284 is runnig python 3.7 so perhaps he can tell us what pytorch version he is running?

MJavadZallaghi, 2024-05-08:
@EtorArza I have tried this command, see the result:
```bash
mjavadzallaghi@mjavadzallaghi-Legion-Pro-5-16IRX9:~/aerial_gym_ws/aerial_gym_reference_governor/aerial_gym/scripts$ conda activate rlgpu
(rlgpu) mjavadzallaghi@mjavadzallaghi-Legion-Pro-5-16IRX9:~/aerial_gym_ws/aerial_gym_reference_governor/aerial_gym/scripts$ conda install pytorch torchvision torchaudio pytorch-cuda=12.1 -c pytorch -c nvidia
Retrieving notices: ...working... done
Channels:
 - pytorch
 - nvidia
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

# All requested packages already installed.

(rlgpu) mjavadzallaghi@mjavadzallaghi-Legion-Pro-5-16IRX9:~/aerial_gym_ws/aerial_gym_reference_governor/aerial_gym/scripts$ 
```
As you mentioned, python version of `rlgpu` is 3.7, while new versions of `pytorch` need 3.8.
And this is my blocking point to find out how I can update the `pythorch` version inside `rlgpu`. 
@EtorArza Did you face this issue? @mihirk284 Any hint?

MJavadZallaghi, 2024-05-08:
@EtorArza Thank you! **I have solved the issue** with your help. 

For the next person who face the issue, I did these steps:

1. Downloaded Isaac Gym SDK (version 4) and extracted the SDK in the home (`~/isaacgym`)
2. Modified the file `~/isaacgym/python/rlgpu_conda_env.yml` to update python version to 3.8:
```yaml
name: rlgpu
channels:
  - pytorch
  - conda-forge
  - defaults
dependencies:
  - python=3.8 #3.7
  #- pytorch=2.3.0 #1.8.1
  #- torchvision=0.9.1
  #- cudatoolkit=12.1 #11.1
  - pyyaml>=5.3.1
  - scipy>=1.5.0
  - tensorboard>=2.2.1
```
4. in the directory `~/isaacgym`, run the command `./create_conda_env_rlgpu.sh` and wait for `rlgpu` conda env installation.
5. activate `rlgpu` env by `conda activate rlgpu`.
6. run command `conda install -c fvcore -c iopath -c conda-forge fvcore iopath`. DO NOT run the command `conda install -c pytorch3d pytorch3d` as it will downgrade some torch related libraries.  `pytorch3d` will be installed in the next steps.
7.  Go to the directory of aerial gym `aerial_gym_simulator/` and run `pip3 install -e .` This step will install `pytorch3d` for you.
8. Got to the example directories of aerial gym and run `python example.py`. You must see this beautiful scene now:
[image]

**For more information:**

- My OS:  Ubuntu 20.04
- My GPU: RTX 4060
- Cuda version: 12.2
```bash
(rlgpu) mjavadzallaghi@mjavadzallaghi-Legion-Pro-5-16IRX9:~/aerial_gym_ws/aerial_gym_reference_governor/aerial_gym/scripts$ nvidia-smi 
Wed May  8 12:01:30 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.171.04             Driver Version: 535.171.04   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4060 ...    Off | 00000000:01:00.0 Off |                  N/A |
| N/A   37C    P4              N/A /  50W |     14MiB /  8188MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1162      G   /usr/lib/xorg/Xorg                            4MiB |
|    0   N/A  N/A      1958      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+
```
- list of the installed packages and their version in the rlgpu env:
```bash
# packages in environment at /home/mjavadzallaghi/anaconda3/envs/rlgpu:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0              pyhd8ed1ab_0    conda-forge
aerial-gym                1.0.0                     dev_0    <develop>
brotli-python             1.1.0            py38h17151c0_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
[... 89 lines cut ...]
six                       1.16.0             pyh6c4a22f_0    conda-forge
sympy                     1.12                     pypi_0    pypi
tabulate                  0.9.0              pyhd8ed1ab_1    conda-forge
tensorboard               2.16.2             pyhd8ed1ab_0    conda-forge
tensorboard-data-server   0.7.0            py38hcdda232_1    conda-forge
termcolor                 2.4.0              pyhd8ed1ab_0    conda-forge
tk                        8.6.13          noxft_h4845f30_101    conda-forge
torch                     2.3.0                    pypi_0    pypi
torchvision               0.18.0                   pypi_0    pypi
tqdm                      4.66.4             pyhd8ed1ab_0    conda-forge
transitions               0.9.0                    pypi_0    pypi
triton                    2.3.0                    pypi_0    pypi
typing-extensions         4.11.0                   pypi_0    pypi
urllib3                   2.2.1              pyhd8ed1ab_0    conda-forge
werkzeug                  3.0.3              pyhd8ed1ab_0    conda-forge
wheel                     0.43.0             pyhd8ed1ab_1    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yacs                      0.1.8              pyhd8ed1ab_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge

```

piratax007, 2024-06-04:
Hi, I followed the steps commented by @MJavadZallaghi but now I received this error:

```
Importing module 'gym_38' (/home/r2d2/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/r2d2/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
AERIAL_GYM_ROOT_DIR /home/r2d2/isaacgym/aerial_gym_simulator
PyTorch version 2.3.0+cu121
Device count 1
/home/r2d2/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/r2d2/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Creating extension directory /home/r2d2/.cache/torch_extensions/py38_cu121/gymtorch...
Emitting ninja build file /home/r2d2/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/2] c++ -MMD -MF gymtorch.o.d -DTORCH_EXTENSION_NAME=gymtorch -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -isystem /home/r2d2/anaconda3/envs/rlgpu/lib/python3.8/site-packages/torch/include -isystem /home/r2d2/anaconda3/envs/rlgpu/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -isystem /home/r2d2/anaconda3/envs/rlgpu/lib/python3.8/site-packages/torch/include/TH -isystem /home/r2d2/anaconda3/envs/rlgpu/lib/python3.8/site-packages/torch/include/THC -isystem /home/r2d2/anaconda3/envs/rlgpu/include/python3.8 -D_GLIBCXX_USE_CXX11_ABI=0 -fPIC -std=c++17 -DTORCH_MAJOR=2 -DTORCH_MINOR=3 -c /home/r2d2/isaacgym/python/isaacgym/_bindings/src/gymtorch/gymtorch.cpp -o gymtorch.o 
[2/2] c++ gymtorch.o -shared -L/home/r2d2/anaconda3/envs/rlgpu/lib/python3.8/site-packages/torch/lib -lc10 -ltorch_cpu -ltorch -ltorch_python -o gymtorch.so
Loading extension module gymtorch...
Traceback (most recent call last):
  File "example.py", line 12, in <module>
    from aerial_gym.envs import *
  File "/home/r2d2/isaacgym/aerial_gym_simulator/aerial_gym/envs/__init__.py", line 9, in <module>
    from .base.aerial_robot  import AerialRobot
  File "/home/r2d2/isaacgym/aerial_gym_simulator/aerial_gym/envs/base/aerial_robot.py", line 17, in <module>
    from isaacgym.torch_utils import *
  File "/home/r2d2/isaacgym/python/isaacgym/torch_utils.py", line 135, in <module>
    def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
  File "/home/r2d2/anaconda3/envs/rlgpu/lib/python3.8/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
```

**For more information**

- OS: Ubuntu 22.04.4 LTS
- GPU: RTX 4090
- CUDA: 12.2
- list of installed packages and verions
```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0              pyhd8ed1ab_0    conda-forge
aerial-gym                1.0.0                     dev_0    <develop>
brotli-python             1.1.0            py38h17151c0_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
ca-certificates           2024.6.2             hbcca054_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1                    pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
filelock                  3.14.0                   pypi_0    pypi
fonttools                 4.53.0                   pypi_0    pypi
freetype                  2.12.1               h267a509_2    conda-forge
fsspec                    2024.6.0                 pypi_0    pypi
fvcore                    0.1.5.post20210915            py38    fvcore
grpcio                    1.62.2           py38h94a1851_0    conda-forge
idna                      3.7                pyhd8ed1ab_0    conda-forge
imageio                   2.34.1                   pypi_0    pypi
importlib-metadata        7.1.0              pyha770c72_0    conda-forge
importlib-resources       6.4.0                    pypi_0    pypi
iopath                    0.1.9                      py38    iopath
isaacgym                  1.0rc4                    dev_0    <develop>
jinja2                    3.1.4                    pypi_0    pypi
kiwisolver                1.4.5                    pypi_0    pypi
lcms2                     2.16                 hb7c19ff_0    conda-forge
[... 67 lines cut ...]
scipy                     1.10.1           py38h59b608b_3    conda-forge
setuptools                70.0.0             pyhd8ed1ab_0    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
[... cut ...]
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yacs                      0.1.8              pyhd8ed1ab_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
```

closed 2024-07-22 by mihirk284