ntnu-arl/aerial_gym_simulator #2: [Request] Extremely simple example with a dynamic obstacle

romesco, 2023-11-15:
Hello!

It would be awesome if you could add a simple example with a dynamic obstacle (that is loaded via assets) and follows some fixed trajectory in a loop!

Thanks for the great repo!

mihirk284, 2023-11-22:
A quick way to do this would be the following:
1. Save a tensor slice corresponding to the object position as a PyTorch tensor
```python
        self.root_states = self.vec_root_tensor[:, 0, :]
        self.root_object_states = self.vec_root_tensor[:, 1:, :]   # this line is added to the existing code
        self.root_positions = self.root_states[..., 0:3]
        self.root_quats = self.root_states[..., 3:7]
        self.root_linvels = self.root_states[..., 7:10]
        self.root_angvels = self.root_states[..., 10:13]
```

**In this case, only a new line defining `self.root_object_states` has been added to the existing code**

2. The next would be to change the tensor at each time step to move the object:
This can be added in the `def step(...)` function  before the environment reset calls in the following manner:

```python
        # update object positions
        pos_param = torch.tensor([self.counter/100.0], device=self.device)
        self.root_object_states[:, :, 0] = torch.sin(pos_param)
        self.root_object_states[:, :, 1] = torch.cos(pos_param)
        self.root_object_states[:, :, 2] = 2.0
        self.gym.set_actor_root_state_tensor(self.sim, self.root_tensor)
```

The above changes lead to a simulation that looks like this:
[image]

While this is a quick and dirty hack to "teleport" objects in the simulation, it can work better if either velocity is written to the object's state. However, continuously updating the state at each time step is not very good for physics realism. So an alternate way would be to apply forces on each object to have it track a position or a trajectory. You can make use of the controllers to also apply forces to the objects and not only the robot :)

mihirk284, 2024-05-08:
Closed issue as resolved

closed 2024-05-08 by mihirk284