ntnu-arl/aerial_gym_simulator #5: Cameras Extraction

mohamedamr13, 2024-03-16:
[image]
How can I extract both the RGB and depth images ? I can only find one type of images saved in  the full_camera_array ( probably the depth image as it is 1D)

mihirk284, 2024-03-21:
Hello, the example files provide examples using depth cameras. 

To use RGB images, please create another tensor array and use the following to access the RGB image:
`camera_image_tensor = self.gym.get_camera_image_gpu_tensor(self.sim, env_handle, cam_handle, gymapi.IMAGE_COLOR)`

This definition can be placed [here](https://github.com/ntnu-arl/aerial_gym_simulator/blob/5cc27b79cbf3caa142d63973faab0280bede5e30/aerial_gym/envs/base/aerial_robot_with_obstacles.py#L193C17-L193C123).

With this you will be able to access the RGB images captured by the camera.

mohamedamr13, 2024-03-23:
> Hello, the example files provide examples using depth cameras.
>
> To use RGB images, please create another tensor array and use the following to access the RGB image: `camera_image_tensor = [...]

Thanks a lot! @mihirk284

closed 2024-03-23 by mohamedamr13