ntnu-arl/aerial_gym_simulator #7: Reward Size Mismatch running RL games play script for a trained checkpoint

mohamedamr13, 2024-04-22:
I am trying to run the rl_games play script to test the policy I have trained, but I got a series of errors in rl_games that I solved till reaching this error. 

[image]

File "/home/user/roshdim1/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/torch_runner.py", line 123, in run_play
    player.run()
  File "/home/user/roshdim1/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/common/player.py", line 332, in run
    cr += r
RuntimeError: The size of tensor a (8) must match the size of tensor b (13) at non-singleton dimension 2

There appears to be a mismatch in the sizes of the cr and the reward returned from the env. Is there a simpler way to run simulation of a trained checkpoint ?

This is the command I ran

python3 runner.py --play --task="quad_with_obstacles" --num_envs=8 --checkpoint=runs/quad_ppo_22-19-01-34/nn/quad_ppo.pth

mohamedamr13, 2024-04-22:
That is the mismatch in the cr rewards and rewards vectors. Is there any reason for this mismatch. 8 is apparently the batch size. 
crshape torch.Size([8]) ]
rshape torch.Size([8, 19, 13])

mohamedamr13, 2024-04-22:
[image]
The problem turned out to be line 185 in the env_step method in the rl_games player script. 
The correct sequence of the step method should be 
        obs, _ , rewards, dones, infos = env[0].step(actions)

Since the step method of the aerial-robot envs returns 
         self.obs_buf, self.privileged_obs_buf, self.rew_buf, self.reset_buf, self.extras
However I am still unsure if the reset_buf coincides with the dones 

is there any pointer for that ? @mihirk284

mihirk284, 2024-07-23:
Hello @mohamedamr13  , a new version has been released for Aerial Gym now with the signals to the RL algorithm confirming to the latest Gymnasium specifications!

mihirk284, 2024-08-03:
Closing as this issue is resolved with the latest version of the simulator

closed 2024-08-03 by mihirk284