ntnu-arl/aerial_gym_simulator #8: multiple drone missions

wagh311, 2024-06-17:
Can this code base be used to train multiple drone missions? Like multiple drone navigation?

mihirk284, 2024-07-01:
It's possible to spawn multiple drones in the same environment.

Currently, in our case, we spawn one robot per environment and one controller per robot. This can however be changed and multiple robots can be spawned and the states of multiple robots can be sent to the controller to act upon simply by reshaping them appropriately. 

Subsequently, the controller outputs can be reshaped to command the appropriate rigid bodies in the same environment. Unfortunately, we will not be able to add support for the same in a short-enough timeframe ourselves so I would like to encourage you to try it out and revert on this issue if you need help :)

wagh311, 2024-07-01:
Thanks for your reply, I will give it a try!

closed 2024-08-03 by mihirk284