ntnu-arl/aerial_gym_simulator #9: Test with action_mean in cleanrl

LjXiii, 2024-07-15:
I want to use action_mean (instead of random sampling) to test the already trained neural network model in CleanRL. The results are very poor, but using the original random sampling method for testing yields very good results. I would like to ask why this might be happening?

The testing code is modified as shown in the uploaded image, using action_mean for testing.
[image]

mihirk284, 2024-07-23:
Hello @LjXiii, I am not sure why the specific RL framework is not yielding desired results. I had tried the same with Sample-Factory and was getting expected results.  Do other frameworks also cause a similar behavior?

mihirk284, 2024-12-11:
Closing this issue.

@LjXiii  Please let me know if this is still not resolved and I will reopen it.

closed 2024-12-11 by mihirk284