ntnu-arl/aerial_gym_simulator #11: Issue with Importing verifiable_learning Module

mychoi97, 2024-07-20:
Hi,

I am currently trying to run the `position_control_example.py` script from the `aerial_gym` package, but I am encountering an import error related to the `verifiable_learning` module. Below is the error message I receive:

Traceback (most recent call last):
File "position_control_example.py", line 1, in <module>
from aerial_gym.utils.logging import CustomLogger
File "/home/mychoi/Research/Quadrotor/aerial_gym_ws/src/aerial_gym_simulator/aerial_gym/init.py", line 6, in <module>
from .task import *
File "/home/mychoi/Research/Quadrotor/aerial_gym_ws/src/aerial_gym_simulator/aerial_gym/task/init.py", line 5, in <module>
from aerial_gym.task.navigation_task.navigation_task import NavigationTask
File "/home/mychoi/Research/Quadrotor/aerial_gym_ws/src/aerial_gym_simulator/aerial_gym/task/navigation_task/navigation_task.py", line 10, in <module>
from aerial_gym.utils.vae.vae_image_encoder import VAEImageEncoder
File "/home/mychoi/Research/Quadrotor/aerial_gym_ws/src/aerial_gym_simulator/aerial_gym/utils/vae/vae_image_encoder.py", line 3, in <module>
from verifiable_learning.DepthToLatent.networks.VAE.vae import VAE
ModuleNotFoundError: No module named 'verifiable_learning'

Could you please provide guidance on how to resolve this issue? Is there a specific repository or installation method I should follow to obtain the `verifiable_learning` module?

Thank you for your assistance.

mihirk284, 2024-07-22:
@mychoi97  thank you for pointing this out. The dependency is now removed and the associated code is absorbed into the Aerial Gym Simulator. Please refer to this commit: [b3b2ce4c81b5c9e297ce2c5ef38fd06d4db1b775](https://github.com/ntnu-arl/aerial_gym_simulator/commit/b3b2ce4c81b5c9e297ce2c5ef38fd06d4db1b775)

You should now be able to run the VAE for the navigation task :)

mychoi97, 2024-07-22:
Thank you for the update!

closed 2024-07-22 by mychoi97