ntnu-arl/aerial_gym_simulator #12: Error Running: bash run_trained_navigation_policy.sh

mb9041, 2024-07-31:
Hello!

I am currently working through the reinforcement learning documentation and encountered an issue while trying to check the performance of the trained model using the script `bash run_trained_navigation_policy.sh`.

I receive the following error message:
`dce_nn_navigation.py: error: unrecognized arguments: --train_dir=/path_to_ws/aerial_gym_simulator/aerial_gym/examples/dce_rl_navigation/selected_network --experiment=selected_network --env=test --obs_key=observations --load_checkpoint_kind=best`

Upon inspecting the get_args() method, I noticed that the arguments being passed in the shell script are not recognized by the script.

Could you please let me know if there is a better example I should start with, or if I am running the script incorrectly? Any guidance would be greatly appreciated.

Thank you!

mihirk284, 2024-07-31:
Hello, this seems to be an issue with the arguments that are being sent to Isaac Gym's agrparser that we use.

Please change the code in isaacgym's `gymutil.py` file (line 337) to:
```python3
args, _ = parser.parse_known_args()
```
This enables the parser to parse known arguments while ignoring those that are not registered with it.

For more information please see the associated section in the [Installation instructions](https://ntnu-arl.github.io/aerial_gym_simulator/2_getting_started/#installation).

mb9041, 2024-07-31:
Thank you so much, this worked!

closed 2024-07-31 by mb9041