ntnu-arl/aerial_gym_simulator #16: 您好，我想询问aerial_gym_simulator的一些使用问题

ncepu666, 2024-09-01:
您好，我是一个纯新手小白，我想做一些基于learning的无人机端到端的避障策略训练，这时我第一次接触，还请您不要嫌弃问题愚蠢。
我想基于您的四旋翼无人机模型（可能会使用雷达），在自己新搭建的环境里去训练，我已经观看了官网的文档，Registries、Assets、Environments、Tasks这些部分我真的看的一头雾水，我需要怎么做才能完成一整个pipline呢？这些接口又该如何去获取呢？有没有类似于单片机最小系统的哪种说法呢？虚心求教——————一位来自中国的学生

ncepu666, 2024-09-01:
我现在超级混乱，又是urdf，又是assets、registry、sim的还有config、envmanger 的 我想自己搭建一片森林 我根本不知道从何入手

mihirk284, 2024-09-04:
Hello, 

There is an example file provided in the code that can help you get started with training depth-based collision avoidance policies in cluttered environments. 

The code for the same can be found and edited to suit your needs [here](https://github.com/ntnu-arl/aerial_gym_simulator/blob/main/aerial_gym/examples/navigation_task_example.py)

In a general case, to just begin trying out the code, you do not need to touch any of the code. You might just need to start training with a chosen framework. For example, with rl_games you can just begin with [this command](https://ntnu-arl.github.io/aerial_gym_simulator/6_rl_training/#train-your-own-navigation-policies):

```python
### Train the navigation policy for the quadrotor with velocity control
python3 runner.py --file=./ppo_aerial_quad_navigation.yaml --num_envs=1024 --headless=True
```

rl_games will be used to train a policy to perform collision avoidance in a cluttered environment. It should take about an hour or two to train on a sufficiently powerful GPU (in headless mode).

Once you are ready and done training you can try iterating on how the environment is set up, how the data is processed and make changes suited to your needs in the `navigation_task` code.

mihirk284, 2024-12-11:
Closing this issue. 

@ncepu666  please let me know if this is still not resolved. I will then reopen it for further clarifications.

closed 2024-12-11 by mihirk284