ntnu-arl/aerial_gym_simulator #19: About dce map

lovelyyoshino, 2024-12-11:
Hello, First of all, this is an excellent piece of work. I would like to inquire about the paper titled "Reinforcement Learning for Collision-free Flight Exploiting Deep Collision Encoding." I couldn't find the training code in this project. Could you please let me know where I can find it or where I can access the open-source code?

mihirk284, 2024-12-11:
Hello, 

Thank you for you kind comments!
We provide the training code as one of the task examples itself. You will be able to train for the policy with the following command:

```python
python3 train_aerialgym.py --env=navigation_task --experiment="<some name for the folder as you like>"
```
This is executed in the `rl_training/sample-factory/aerialgym_examples` folder.

This task trains a quadrotor platform named "lmf2" with a depth camera similar to realsense d455 (downsampled deth image) mounted in the front, simulated in a room-like environment with floating obstacles (static). The code for the task implementation is [here](https://github.com/ntnu-arl/aerial_gym_simulator/blob/main/aerial_gym/task/navigation_task/navigation_task.py), with the DCE latent code being inferred in the code [here](https://github.com/ntnu-arl/aerial_gym_simulator/blob/7a77ada2a55a0804ed978d2dac7babb2c2dcd55a/aerial_gym/task/navigation_task/navigation_task.py#L278).  The encoder module is used from [this class](https://github.com/ntnu-arl/aerial_gym_simulator/blob/main/aerial_gym/utils/vae/vae_image_encoder.py#L17).

We send back the latent code  (and not the depth image observations) to the RL policy for training. In my experience, the policy should reach about curriculum level of 40 (6 walls + 34 obstacles) in the environment  in less than 2 hours on an RTX 3090 when trained with 1024 environments.

Please let me know if you have any trouble running the code :)

lovelyyoshino, 2024-12-12:
Thks, I will come and read it carefully

closed 2024-12-12 by lovelyyoshino