ntnu-arl/aerial_gym_simulator #20: Error in getting images

pipixiaCommander, 2024-12-26:
Hi,
I have just installed aerialgym and ran the `save_camera_stream.py `example, but encountered a problem with generating images, which I am at a complete loss about. I would greatly appreciate your help, thanks!

Here is my detailed runtime information:

Importing module 'gym_38' (/home/admin/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/admin/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 1.13.1
Device count 1
/home/admin/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/admin/.cache/torch_extensions/py38_cu117 as PyTorch extensions root...
Emitting ninja build file /home/admin/.cache/torch_extensions/py38_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Warp 1.0.0-beta.5 initialized:
   CUDA Toolkit: 11.5, Driver: 12.5
   Devices:
     "cpu"    | x86_64
     "cuda:0" | NVIDIA GeForce RTX 4060 Ti (sm_89)
   Kernel cache: /home/admin/.cache/warp/1.0.0-beta.5
[3626 ms][__main__] - WARNING : 

Environment to save a depth/range and segmentation image.

 (save_camera_stream.py:12)
[3627 ms][env_manager] - INFO : Populating environments. (env_manager.py:73)
[3627 ms][env_manager] - INFO : Creating simulation instance. (env_manager.py:87)
[3627 ms][env_manager] - INFO : Instantiating IGE object. (env_manager.py:88)
[3628 ms][IsaacGymEnvManager] - INFO : Creating Isaac Gym Environment (IGE_env_manager.py:41)
[3628 ms][IsaacGymEnvManager] - INFO : Acquiring gym object (IGE_env_manager.py:73)
[3628 ms][IsaacGymEnvManager] - INFO : Acquired gym object (IGE_env_manager.py:75)
[3634 ms][IsaacGymEnvManager] - INFO : Fixing devices (IGE_env_manager.py:89)
[3634 ms][IsaacGymEnvManager] - INFO : Sim Device type: cuda, Sim Device ID: 0 (IGE_env_manager.py:96)
[3634 ms][IsaacGymEnvManager] - INFO : Graphics Device ID: 0 (IGE_env_manager.py:110)
[3634 ms][IsaacGymEnvManager] - INFO : Creating Isaac Gym Simulation Object (IGE_env_manager.py:111)
[3634 ms][IsaacGymEnvManager] - WARNING : If you have set the CUDA_VISIBLE_DEVICES environment variable, please ensure that you set it
to a particular one that works for your system to use the viewer or Isaac Gym cameras.
If you want to run parallel simulations on multiple GPUs with camera sensors,
please disable Isaac Gym and use warp (by setting use_warp=True), set the viewer to headless. (IGE_env_manager.py:118)
[3634 ms][IsaacGymEnvManager] - WARNING : If you see a segfault in the next lines, it is because of the discrepancy between the CUDA device and the graphics device.
Please ensure that the CUDA device and the graphics device are the same. (IGE_env_manager.py:123)
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
[6406 ms][IsaacGymEnvManager] - INFO : Created Isaac Gym Simulation Object (IGE_env_manager.py:127)
[6406 ms][IsaacGymEnvManager] - INFO : Created Isaac Gym Environment (IGE_env_manager.py:43)
[15771 ms][env_manager] - INFO : IGE object instantiated. (env_manager.py:109)
[15771 ms][env_manager] - INFO : Creating warp environment. (env_manager.py:112)
[15771 ms][env_manager] - INFO : Warp environment created. (env_manager.py:114)
[15771 ms][env_manager] - INFO : Creating robot manager. (env_manager.py:118)
[15771 ms][BaseRobot] - INFO : [DONE] Initializing controller (base_robot.py:26)
[15771 ms][BaseRobot] - INFO : Initializing controller lee_velocity_control (base_robot.py:29)
[15771 ms][base_multirotor] - WARNING : Creating 2 multirotors. (base_multirotor.py:32)
[15772 ms][env_manager] - INFO : [DONE] Creating robot manager. (env_manager.py:123)
[15772 ms][env_manager] - INFO : [DONE] Creating simulation instance. (env_manager.py:125)
[15772 ms][asset_loader] - INFO : Loading asset: quad.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15772 ms][asset_loader] - INFO : Loading asset: panel.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15774 ms][asset_loader] - INFO : Loading asset: 0_5_x_0_5_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15775 ms][asset_loader] - INFO : Loading asset: small_cube.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15776 ms][asset_loader] - INFO : Loading asset: 1_x_1_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15777 ms][asset_loader] - INFO : Loading asset: cuboidal_rod.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15778 ms][asset_loader] - INFO : Loading asset: left_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15778 ms][asset_loader] - INFO : Loading asset: right_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15779 ms][asset_loader] - INFO : Loading asset: back_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15780 ms][asset_loader] - INFO : Loading asset: front_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[15781 ms][asset_loader] - INFO : Loading asset: bottom_wall.urdf for the first time. Next use of this asset will be via the asset buffer. (asset_loader.py:71)
[... 120 lines cut ...]
F: Toggle Camera Follow
P: Toggle Camera Follow Type
R: Reset All Environments
UP: Switch Target Environment Up
DOWN: Switch Target Environment Down
SPACE: Pause Simulation
 (IGE_viewer_control.py:153)
[11200 ms][IsaacGymEnvManager] - INFO : Created viewer (IGE_env_manager.py:423)
*** Can't create empty tensor
[12160 ms][asset_manager] - WARNING : Number of obstacles to be kept in the environment: 9 (asset_manager.py:32)
WARNING: allocation matrix is not full rank. Rank: 4
/home/admin/workspaces/aerial_gym_ws/src/aerial_gym_simulator/aerial_gym/control/motor_model.py:38: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  torch.tensor(self.min_thrust, device=self.device, dtype=torch.float32).expand(
/home/admin/workspaces/aerial_gym_ws/src/aerial_gym_simulator/aerial_gym/control/motor_model.py:41: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  torch.tensor(self.max_thrust, device=self.device, dtype=torch.float32).expand(
[13408 ms][control_allocation] - WARNING : Control allocation does not account for actuator limits. This leads to suboptimal allocation (control_allocation.py:48)
[15392 ms][__main__] - ERROR : Error in getting images (save_camera_stream.py:83)
[15392 ms][__main__] - ERROR : Seems like the image tensors have not been created yet. (save_camera_stream.py:84)
[15392 ms][__main__] - ERROR : This is likely due to absence of a functional camera in the environment (save_camera_stream.py:85)
Traceback (most recent call last):
  File "save_camera_stream.py", line 86, in <module>
    raise e
  File "save_camera_stream.py", line 79, in <module>
    255.0 * env_manager.global_tensor_dict["depth_range_pixels"][0, 0].cpu().numpy()
KeyError: 'depth_range_pixels'

mihirk284, 2024-12-27:
Hello, 
Thank you for opening an issue for this error. 

A quick fix would be to set `enable_camera = True` [in this file](https://github.com/ntnu-arl/aerial_gym_simulator/blob/bb9d59869913c9f32e1ddb129fee098616a98d7c/aerial_gym/config/robot_config/base_quad_config.py#L57).

Apologies for this error. A fix will be implemented soon that will allow running all examples without an error.

pipixiaCommander, 2024-12-30:
Thks,it works.

closed 2024-12-30 by pipixiaCommander