ntnu-arl/aerial_gym_simulator #28: dce_rl_navigation environment crash

krank10, 2025-01-31:
Hello,
    thank you for your wonderful and inspiring work. When running the example examples/dce_rl_navigation (by executing bash run_trained_navigation_policy.sh), the environment crashed in an extremely short time (only 1 time step). how can I solve this problem?

Resetting environments (tensor([7], device='cuda:0'),) due to Timeout
Resetting environments (tensor([13], device='cuda:0'),) due to Timeout
Resetting environments (tensor([4], device='cuda:0'),) due to Timeout
Resetting environments (tensor([11], device='cuda:0'),) due to Termination
Resetting environments (tensor([8], device='cuda:0'),) due to Timeout
Resetting environments (tensor([3], device='cuda:0'),) due to Termination
Resetting environments (tensor([0], device='cuda:0'),) due to Timeout
Resetting environments (tensor([13], device='cuda:0'),) due to Termination
Resetting environments (tensor([1], device='cuda:0'),) due to Timeout
Resetting environments (tensor([2], device='cuda:0'),) due to Termination

    best.

krank10, 2025-01-31:
and I can't see any  aerial robots, just the environment like this: 

[image]

mihirk284, 2025-02-02:
Hello,

Thank you for opening an issue. Based on the output from the scripts in the first message of the issue above, it is unclear how the environments are crashing on the first timestep. Based on the output statements above, it seems like the robots are indeed navigating the environment and some of them are reset due to a Timeout (which means that the robots are exceeding the maximum episode lengths for the task). 

Perhaps if you can share more information or upload a GIF to indicate the same I can try to replicate it on my end. 

The environments here are essentially room-like with floating objects inside them. They have floating objects inside them. You can toggle the view such that the viewer camera follows the robot or is static in the environment by pressing the "F" key on your keyboard. If you press "P" key when the robot is in the follow mode, it will switch to a different camera view that aligns the camera pose with the global frame's heading while still following the robot. In this mode, you can press "UP" and "DOWN" keys to toggle the environment of interest such that you can follow a different robot.

Please do let me know if this does not work for you. If the issue with the resets persists, please let me know with more information and I can try to replicate it on my end.

Best,

krank10, 2025-02-03:
Thank you for your valuable and detailed guidance! It's been enlightening to see the drone's navigation tasks by switching to a different camera view. However, I'm still encountering some issues where environments are terminated or timeout. Here are the terminal logs for your reference. Is this behavior expected?

[terminal logs.md](https://github.com/user-attachments/files/18639682/terminal.logs.md)

mihirk284, 2025-02-03:
If the line you refer to is:

```
[2025-02-03 16:36:13,738][3133158] Loading state from checkpoint /home/eva/rl/aerial_gym_simulator-main/aerial_gym/examples/dce_rl_navigation/selected_network/selected_network/checkpoint_p0/best_000052096_26673152_reward_1333.322.pth...
[5009 ms][navigation_task] - CRITICAL : Crash is happening too soon. (navigation_task.py:195)
[5009 ms][navigation_task] - CRITICAL : Envs crashing too soon: tensor([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15],
       device='cuda:0') (navigation_task.py:196)
[5010 ms][navigation_task] - CRITICAL : Time at crash: tensor([1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1], device='cuda:0',
```

Then this is perfectly fine. All environments reset at the beginning of running the code. If this shows up later, then this is a problem.

mihirk284, 2025-02-04:
If you refer to the logs where the environments are either being terminated or experiencing a timeout, this is to be expected.

Since each episode is having a definite time limit, if the robot is collision-free for the entire limit of time, the environment resets and starts a new episode (this is good thing since the robot has not collided, and is probably hovering at the target location)

If the environment resets due to termination - this means that the robot has suffered a crash and the environment is reset  for a new episode. This is undesirable outcome (but expected) since the policy does not ensure 100% collision-free navigation. 

Moreover, this policy was trained on a previous version of the simulator and is now deployed with the latest release. Unfortunately, this means that there are some inconsistencies in the robot dynamics and how the environment is composed, which slightly reduces the performance compared to what was shown in the ICRA paper. We will be releasing a new policy trained on this version of the simulator soon that has been verified to navigate at much higher speeds in the environment and requires a lesser training time. 

With you question regarding the print statements, these are normal and expected. It would be of concern though if the robots are actively trying to collide against the obstacles instead of trying to avoid them, which you can verify visually through the camera controls.

krank10, 2025-02-05:
Thank you very much for your timely and professional response! Your detailed explanation has been of immense help.

mihirk284, 2025-02-06:
Thank you!

I'm closing the issue now. Please feel free to reopen it if you have any further questions regarding the same script or open a new issue if you have any question regarding other features of the repository

closed 2025-02-06 by mihirk284