ntnu-arl/aerial_gym_simulator #30: Segmentation fault when installing the aerial_gym_simulator

ecolihjy, 2025-02-22:
亲爱的作者您好，我是aerialgym的粉丝，在浏览量aerialgym之后，我想要基于此建立一套属于自己的无人机飞行环境，以便与脑机接口系统相结合，做强化学习相关任务。
我在按照isaacgym官方教程安装好isaacgym之后，里面的实例我能运行（例如joint_monkey.py），但是在按照aerialgym的installation教程配置完Aerial Gym Simulator之后出现了以下错误：

[Error] [carb.gym.plugin] cudaImportExternalMemory failed on rgbImage buffer with error 801
Segmentation fault

请问您有好的解决方法或者建议吗，我将不胜感激

mihirk284, 2025-02-24:
Hello, thank you for opening an issue.

Can you try out other examples and let me know if they work? For example `position_control_example.py`?

mihirk284, 2025-03-05:
Hello, 

The examples provided natively with Isaac Gym are using the CPU pipeline by default. Can you turn this off, and force them to use the GPU pipeline and see if this is still happening for those?

If the same issue happens on using GPU pipeline with Isaac Gym, then there might be a driver version problem. If you have ensured that the GPU pipeline works properly with the native Isaac Gym examples, and this issue still persists with Aerial Gym, I would recommend the following steps:

Some of the reasons are:
1) Perhaps the LD_LIBRARY_PATH might be set incorrectly. Please refer to the documentation (installation / troubleshooting section) for Isaac Gym for setting this.
2) NVIDIA GPU driver inconsistencies - Please check if the NVIDIA Driver would be correct. I currently have 550.107.02 and that seems to work on my laptop with Ubuntu 22.04
3) Check what GPU device the simulation is being run on. Please check that the GPU device that has been set is one that is connected to a display. Please also check if this issue still persists on using Aerial Gym in headless mode.

Please let me know if this does not work and I can try to help you further :)

ecolihjy, 2025-03-17:
Thank you for your help. First of all, there is no nvidia_icd. json file in my icd. d folder:

[image]

Secondly, the driver has already been set up and the GPU has been forced to be used

Third, it operated normally in headless mode。

Can you provide some more suggestions to solve the graphics rendering problem。

mihirk284, 2025-03-22:
If it operates regularly in headless mode but fails on trying to spawn a rendering window, I suspect it is either the GPU device selection configuration for the system or some vulkan configuration. I think this a system-specific issue. 

I can tell you that in my case this issue always happens when I run the simulation in non-headless mode on a GPU that is not connected to a display. Headless works in this case, but Isaac Gym-based rendering fails (both for the viewer window and simulated camera sensors). The solution is running the simulation on a GPU connected to a display and this resolves.
Hope this helps.

ecolihjy, 2025-03-24:
Thanks for your reply! 
I tried it once as you suggested, and it was indeed running on the same GPU, which was also connected to the monitor. However, there was still a segmentation fault issue and the following text appeared:

[image]

Do you think this could be a problem with Vulkan? Do you think it's necessary to reinstall an adapted Vulkan？

mihirk284, 2025-04-08:
I think so. Does `vulkaninfo` command on the shell produce a valid result regarding your devices? If not, then there must be a problem with Vulkan on your machine

ecolihjy, 2025-04-13:
Thanks for your reply!
Vulkaninfo can generate information normally.Perhaps I will try reinstalling Vulkan to resolve the issue
thank again for your patiently answer

ecolihjy, 2025-04-13:
But I have another question, how does the code implement Wandb recording. More specifically, where does wandb.log record information such as rewards and losses？

mihirk284, 2025-04-13:
The code itself does not interface with wandb. The learning framework `rl_games` and `sample_factory` integrates directly with wandb. In case you want to log something particular from the environment or the rollouts that the learning frameworks does not already log, make use of the "extras" of "infos" value that is returned along with the state observations, rewards, truncations and terminations from aerial gym. Both these learning frameworks log everything that on the "extras" or "infos" variable to wandb internally  :)

ecolihjy, 2025-04-14:
Thank you for your answer. I will give it a try :)

mihirk284, 2025-04-14:
I'll close this issue for the time being.  Feel free to reopen the issue if the problem still persists and if you are not able to get it to work

closed 2025-04-14 by mihirk284