ntnu-arl/aerial_gym_simulator #31: environmental configuration

zhw970623, 2025-03-03:
When testing the navigation strategy, I hope that the target point and environmental obstacles will not be changed when each episode is reset. Can this be set in the configuration file throughout the simulator?

mihirk284, 2025-03-03:
Hello, thank you for opening an issue.

I do not currently have this as a configuration. I would recommend that you add this within the task configuration file and read from this to set the goal and start position. This should be fairly easy to do either via the configuration file or via hard-coding changes in the task definition for the navigation task.

Best,
Mihir

zhw970623, 2025-03-04:
Thank you very much for your answer, but I have another question. Can I cancel the randomization of the position and size of the environment (assets such as obstacles) every time I reset it when testing the strategy?

mihirk284, 2025-03-05:
Currently there is no configuration flag to prevent reset of environments. You can stop the reset process by commenting the lines here https://github.com/ntnu-arl/aerial_gym_simulator/blob/85114e7514ed0f3f8cfecfa23c40dad106c528a1/aerial_gym/env_manager/env_manager.py#L282

I would propose making a function similar to this that does not reset the environment by commenting the following lines:

```python
    def reset_robot_only_idx(self, env_ids=None):
        """
        This function resets the environment for the given environment indices.
        """
        # first reset the Isaac Gym environment since that determines the environment bounds
        # then reset the asset managet that respositions assets within the environment
        # then reset the warp environment if it is being used that reads the state tensors from the assets and transforms meshes
        # finally reset the robot manager that resets the robot state tensors and the sensors
        # logger.debug(f"Resetting environments {env_ids}.")
        # self.IGE_env.reset_idx(env_ids)
        # self.asset_manager.reset_idx(env_ids, self.global_tensor_dict["num_obstacles_in_env"])
        # if self.cfg.env.use_warp:
        #     self.warp_env.reset_idx(env_ids)
        self.robot_manager.reset_idx(env_ids)
        self.IGE_env.write_to_sim()
        self.sim_steps[env_ids] = 0
```

And call this new function that you make that resets everything else but the environment and object states during the run.

This ensures that the objects in the environment are not reset but the robot is reset to the desired position.

zhw970623, 2025-03-12:
Thank you very much for your excellent contribution. When using the simulator, I found a problem that confused me. What determines the number of obstacles loaded into an env? I saw many factors in the code. What determines the num_obstacles_per_env value passed in by the reset_idx method in the AssetManager class? Where should I configure it?

mihirk284, 2025-03-22:
This would be a task-defined parameter. You can set up a curriculum that specifies the number of obstacles you want the agent to deal with in that round of resets.

closed 2025-03-24 by zhw970623