ntnu-arl/aerial_gym_simulator #33: Technical Exchange

zhw970623, 2025-03-16:
What is the relationship between the `num_assets `attribute in the asset class in```
 aerial_gym_simulator/aerial_gym/config/asset_config
/env_object_config.py
``` and the number of obstacles loaded into the environment?

mihirk284, 2025-03-22:
Thank you for opening an issue.

I will answer the questions pointweise:
1) In the simulator, we are only able to load a fixed number of objects per environment. This needs to be done prior to the simulation actually executing. When we need a variable amount of objects in the robot's vicinity, we spawn a large number of objects, select some objects to be kept near the robot in it's local environment while the others are placed extremely far away (away from sight) so that the robot does not have to deal with them. The num_assets in the asset_class determines the number of objects that are to be spawned in each environment. However, when you reset an environment you can specify how many objects you would like to be in the robot's surroundings by specifying the number alongwith the list of environments that are to be reset. 

Note that the keep_in_env flag being set to true overrides the above behavior and all assets of that class are placed in the robot's vicinity.

2) In the case of the navigation task, the timesteps that are simulated are `task_config.episode_len_steps * env_config.num_physics_steps_per_env_step_mean * sim.dt`

Here, the velocity controller runs at 100Hz (based on sim.dt) but the update to the setpoints via the RL policy happens at 10Hz (this is determined by the `env_config.num_physics_steps_per_env_step_mean` parameter in the env_config file). This allows closed loop-velocity control at a higher rate and therefore also allows an update to the setpoint at a reasonable rate for the RL policy. With this calculation, the 100 steps in the task actually amount to 10 seconds worth of real-world time since the task runs for 100 steps (with each step involving 10 physics steps).

Please feel free to ask for further clarifications or further information if this is unclear or confusing.

Best,
Mihir

zhw970623, 2025-03-23:
Thank you very much for your patience in answering. Regarding the first question (the obstacle setting problem), I still have a doubt. According to what you mean, task_config.current_level specifies the number of obstacles loaded near the agent. However, the prerequisite for placing task_config.current_level obstacles around the agent is that at least current_level obstacles have been initialized during initialization (that is, in env_object_config.py, the combination of num_assets of the classes configured in the environment must be greater than or equal to current_level)

For example
The obstacles configured in the environment are asset_state_params and object_asset_params

Case 1
```
asset_state_params.num_assets = 3.
asset_state_params.keep_in_env = True

object_asset_params=5.
object_asset_params.keep_in_env = False

task_config.current_level=15.
```

Then the obstacles actually loaded near the agent are min[(3 + 5 = 8), 15] = 8.

Case 2
```
asset_state_params.num_assets = 3.
asset_state_params.keep_in_env = True

object_asset_params = 50.
object_asset_params.keep_in_env = False

task_config.current_level = 15.
```
Then the number of obstacles actually loaded near the agent is min[(3 + 50 = 53), 15] = 15.
There are 3 assets and 12 objects.

That is, the number of obstacles you want to place near the agent is at most the sum of all num_assets

mihirk284, 2025-04-08:
Yes, you have correctly pointed out what happens. Note that current_level is a dynamic term. This means that this may change throughout duration of execution. Meaning that if an environment is reset at a given point in time, the number of objects that will be placed in that environment at that point in time (when the reset happens) will be based on the `current_level` variable.

One point to avoid confusion, `asset_state_params` is a parent class of all the specific classes. So for the purpose of this answer I will assume that we are talking about two object parameter classes `panel_asset_params` (instead of `asset_state_params`) and `object_asset_params`.

So in case 1) the total obstacles spawned in the environment are less than the number of objects you want to keep around the robot. So it will keep the lower number of obstacles around based on this line in code: (https://github.com/ntnu-arl/aerial_gym_simulator/blob/cfa69c05f902e1ce064a84ecac0b1e76f5c2805b/aerial_gym/env_manager/asset_manager.py#L52)

In case 2) the number of obstacles that are required (a that point in time based on the value of `current_level` variable at that time) near the robot are 15. But the environment has already loaded 53 obstacles, it is a matter of selection as to which of the 53 obstacles are to be placed near the robot. In this case, the first obstacle type `panel_asset_params` (instead of `asset_state_params` based on your comment) has the `keep_in_env=True`, this overrides the number. So ALL of these will be kept near the robot. So 3 out of 3 are all kept near the robot. From the remaining 50 obstacles, 12 more can be kept near the robot to make sure we reach the number of 15 required obstacles. Therefore, the first 12 are randomly sampled and kept near the robot, (again it does not matter if it is the first 12 or any 12 since we load object instances of the same class at random). The rest of the 38 objects are then put far away (https://github.com/ntnu-arl/aerial_gym_simulator/blob/cfa69c05f902e1ce064a84ecac0b1e76f5c2805b/aerial_gym/env_manager/asset_manager.py#L71).

Please let me know if I can clarify anything in more detail

zhw970623, 2025-04-09:
You explained it very clearly, thank you very much！！！

closed 2025-04-09 by zhw970623