ntnu-arl/aerial_gym_simulator #37: Mismatch Between Paper and Code Results

giangdao1402, 2025-03-31:
Hi @mihirk284 
Thank for your excellence work !
I have a question regarding the difference between your paper and your implementation
The paper "**Reinforcement Learning for Collision-free Flight Exploiting Deep Collision Encoding**" mentioned generating collision image to train VAE model
But in this repo, i can not find the implementation of that task
Do you implement it in this repo ?
Thank you !

mihirk284, 2025-03-31:
Indeed the paper mentioned the use of the collision image to train a collision encoder.  We have currently only provided the trained encoder within this repository to support inference. 

The task that uses the Deep Collision Encoder for navigation can be found here: https://github.com/ntnu-arl/aerial_gym_simulator/blob/main/aerial_gym/task/navigation_task/navigation_task.py

I will update this thread when we open-source the collision image generating pipeline.

mihirk284, 2025-05-08:
Hello, 

We have uploaded the implementation to generate collision images from depth images. This is the code corresponding to the work in "Reinforcement Learning for Collision-free Flight Exploiting Deep Collision Encoding"

The script can be found here: https://github.com/ntnu-arl/aerial_gym_simulator/blob/main/aerial_gym/utils/collision_image_generator.py

giangdao1402, 2025-05-09:
Thank you for the public of the code !

closed 2025-05-09 by giangdao1402