ntnu-arl/aerial_gym_simulator #41: Rendering error

ecolihjy, 2025-04-24:
Hello author!
I encountered rendering issues while using the aerial gym.When calling the isaacgym graphical interface, the rendering is incomplete, as shown in the following figure, with many black vertical stripes appearing. How should this be resolved?

[image]

mihirk284, 2025-05-08:
Hello, this appears to be an issue with the system's graphics driver or the GPU configuration. I have not seen this error before. Does changing the driver fix this?

ecolihjy, 2025-05-12:
After changing the Nvidia driver (whether upgrading or downgrading), the problem remains unresolved.
Besides, I found that when I use vkcube for testing, I am using llvmpipe instead of Geforce3090 GPU, as shown in the following picture. What could be the reason for this? How can I correctly call Geforce3090 GPU?

[image]

ecolihjy, 2025-07-10:
It's okay, thank you very much. I'll come back to ask you if there's any trouble in the future